arduino/ArduinoCore-sam #24: enabling Tone for Due

agdl, 2017-02-08:
_From @HanaJin on October 13, 2014 21:17_

Hi Arduino Team, 

Is there a plan or timeline in mind to enable Tone for Due board in future releases?

_Copied from original issue: arduino/Arduino#2362_

agdl, 2017-02-08:
_From @q2dg on August 8, 2015 13:20_

Well, a solution seems to be here (http://forum.arduino.cc/index.php?topic=136500.0) .
Could you add that piece of code to official tone() function, plis??

...And, on the other hand, what about tone() implementation in Zero, Gemma (#3519) ...??? API must be consistent between boards!!!

agdl, 2017-02-08:
_From @aethaniel on August 10, 2015 13:11_

@q2dg, tone() is implemented on Zero: https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/Tone.cpp

agdl, 2017-02-08:
_From @q2dg on August 10, 2015 16:52_

@aethaniel Ah, oook, thanks! (and sorry!). 
Anyway, clarificating the documentation to specify the diferent API's functionality scope among the diferent boards (if there is any) is fundamental. But that's for another issue.

agdl, 2017-02-08:
_From @q2dg on August 23, 2015 16:44_

Well, I only wanted to point that problem with tone() and Due still remains...

agdl, 2017-02-08:
_From @shiftleftplusone on October 21, 2015 7:37_

I don't see how it now should work for the Due - can you provide a example source code please?

agdl, 2017-02-08:
_From @shiftleftplusone on December 6, 2015 11:22_

OMG, the TOP dated from October 2014  !!
why hadn't it been fixed yet ?!?

agdl, 2017-02-08:
_From @aethaniel on December 6, 2015 23:23_

I have a question to answer yours as you seem to be very demanding on various projects:
Why don't you provide any patch to help on missing features, Helmutt?

## 

Envoyé de mon téléphone Android avec K-9 Mail. Excusez la brièveté.

agdl, 2017-02-08:
_From @shiftleftplusone on December 7, 2015 14:6_

Perhaps you miss something decisive:
I am a user and a customer,  not a manufacturer and not a developer!

Buying a 60 EUR Arduino I simply expect it to work, like I expect it when buying a washing machine or a TV.

Or do you buy either one and then when realizing that there is no water pump inside or no tone on TV 
and thus complaining about those issues at your seller, getting to hear:

"I have a question to answer yours as you seem to be very demanding on various projects: Why don't you provide any patch to help on missing features, aethaniel " ?!?

agdl, 2017-02-08:
> I am a user, not a constructer.
> Buying a Arduino I seemply expect it to work, like I expect buying a
> washing machine or a TV.
> Or do you buy either one and then having no water pump or no tone on
> TV,
> and when complaining those issues at your seller getting to hear
> "I have a question to answer yours as you seem to be very demanding on
> various projects: Why don't you provide any patch to help on missing
> features,aethaniel " ?!?
>
> ---

_From @aethaniel on December 7, 2015 14:30_

Hi,

Like many people, I personally think there is a delta between finished manufactured products and DIY Community boards, even very well branded.

Cheers,

T.

## 

Envoyé de mon téléphone Android avec K-9 Mail. Excusez la brièveté.

agdl, 2017-02-08:
_From @shiftleftplusone on December 7, 2015 14:36_

you may personally think what you wish, but to me a Arduino Due is a finished manufactured product which is simply expected to work by it's libs!

agdl, 2017-02-08:
_From @NicoHood on December 7, 2015 16:36_

He is partly right.

Not everyone is able to patch those things (especially timer related stuff) hisself without reading up on this for hours. Even I do not know where to start, not even on avr. So he is probably right. And tone() is a basic API feature that should be added.

DYI is always incomplete and not perfect. Arduino is the best example, even though it tries to make stuff simple. But time passes, version changes, new boards come and not enough people are working on this. And the Arduino community which just uses the boards will never attempt to fix this. This is hard core.

So yeah, I'd vote for @VogonJeltz that someone fixes this, as its a basic feature like digitalWrite() etc. However I think @VogonJeltz should try to post a bit less aggressive here. Things wont change if you force them to do.

agdl, 2017-02-08:
_From @shiftleftplusone on December 7, 2015 18:12_

I agree, perhaps my posts partially might appear to be sort of harsh - if one didn't know te story before.

I am always  used to ask friendly and politely the first time, but if repetively there comes no answer at all or something like "help yourself, don't expect others to do your homework" and the topic immediately gets closed  then the limit of my politeness has been quickly reached.

Finally this tone issue is pending since the Due has been released, and even in this repository this issue has been successfully ignored since more than one year, nothing just silence in the forest. When would it be time then to slowly getting impatient?

agdl, 2017-02-08:
_From @shiftleftplusone on December 7, 2015 18:18_

BTW, if it's really timer-related stuff (no idea, I don't understand enough about it):
why note set it up to DueTImer by attaching a free interrupt?

Timer.getAvailable().attachInterrupt(duetone).start(); 

https://github.com/ivanseidel/DueTimer/blob/master/examples/AvaliableTimer/AvaliableTimer.pde

thinkyhead, 2018-02-25:
We just did an ad hoc implementation for MarlinFirmware. It's quite small and –dare I say– cute.

EKersting, 2020-06-18:
Anything new on using tone() with the DUE?

RedRussianBear, 2020-06-26:
I've written an implementation and just submitted a pull request, would love to see it be incorporated so folks can finally use this feature. #108

tigoe, 2020-12-23:
Had a report from one of my former students about this issue too, would love to see it completed and closed.

aliphys, 2024-05-22:
I noticed that the `compile-sketches.yml` workflow in the [arduino/arduino-examples](https://github.com/arduino/arduino-examples/) repository mentions this issue at [line 184](https://github.com/arduino/arduino-examples/blob/5d991a2ce69d22e8b21a1ee89ec715f093d45276/.github/workflows/compile-examples.yml#L188). Any updates on this issue?
```
          - fqbn: arduino:sam:arduino_due_x
            usb: true
            serial1: true
            starter-kit: false
            # Bug report: https://github.com/arduino/ArduinoCore-sam/issues/24
            # Change the value to true once it is fixed.
            tone: false
            a5: true
```