arduino/ArduinoCore-sam #30: Using pins 6-9 causes the Due to crash when using the Ethernet R3 shield and WebServer example

agdl, 2017-02-08:
_From @SeanDS on August 13, 2013 22:12_

I'm using the Arduino Ethernet R3 shield with a Due board. The Ethernet shield uses the ICSP connector to communicate with the Due via the SPI socket on the Due, so it frees up pins 11, 12 and 13 for general I/O. Pin 10 is still used for chip select / slave select.

When I use the Ethernet and SPI libraries to set up a simple web server (basically the WebServer example but with an added function to set pins 2-13, excluding 10, to high, when a specific GET command is issued), setting any of pins 6-9 to high (via analogWrite(4095)) causes the web server to become unresponsive.

_Copied from original issue: arduino/Arduino#1539_

agdl, 2017-02-08:
_From @SeanDS on August 13, 2013 22:36_

Ok, I've experimented further and found that the problem goes away when I use analogWrite(x) where x is between 0 and 255. I guess this isn't a serious bug then, but it's still a bug in my eyes. It's still possible to use analogWrite(4095) for pins 2-5 and 11-13 and not have any issues, but when you use analogWrite(4095) for any of pins 6-9 while using the Ethernet library you end up with issues. Very strange.

SeanDS, 2017-02-08:
Yup, pretty strange indeed! Thanks for looking into this, even 4 years after :+1: