arduino/ArduinoCore-sam #45: Flash Page Error in Arduino Due while programming on programming port

pdborhad, 2017-10-15:
Hi,

I am using Arduino due in my project. The project has 14 volt battery power supply. The custom board has power supply unit that converts from 14 volts to 5 volts which goes to arduino due Vin pin. 

However, most of the time I am getting the following error after which I can not use the Arduino board. Please help me how can we avoid the error. This was my 5th board and do not want to bye once in a week. 

I also tried uploading the blink code, but same error. Please find the error below. 

Sketch uses 22124 bytes (4%) of program storage space. Maximum is 524288 bytes.
Atmel SMART device 0x285e0a60 found
Erase flash
done in 0.032 seconds

Write 24392 bytes to flash (96 pages)

[                              ] 0% (0/96 pages)
Flash page is locked
An error occurred while uploading the sketch

facchinm, 2017-10-16:
HI @pdborhad ,
as far as I know, the VIN pin accepts any voltage between 7 and 12 Volts (because it regulates it back to 5V internally). So powering the board with 5V via VIN is going to leave in in a bogus status (probably).
Are you able to reprogram it once remove from the carrier? Have you also tried pressing the "FLASH erase" button manually?

pdborhad, 2017-10-16:
Hi, 

Today again one more board went bad. 

I tried pressing flash button. It programmed when press the flash button, but It did not work. No serial comments can be seen on serial window. This was my 6th board. 
Today I disconnected the external power supply before programming the Due board. Still it went bad. 

Any work around or something that how to use this board while programming? Cant we return these board and ask for new one?

Regards,
Prashant

facchinm, 2017-10-17:
Hi @pdborhad ,
you should contact the support team for replacements, but I'm quite confident that a 100% failure rate on 6 boards is something we never experienced, so it must be generated somehow by your setup. 
The `flash page is locked` error could also be due to a bogus voltage being applied to certain voltage rails, so maybe testing the power rails and providing them in the report could speed up the replacement.

pdborhad, 2017-10-30:
Hi,

I apologize for not posted recently. However, I was checking my code for any issues in flash memory programming. I want to let you know that my Vin voltage is not 5 volts, but it is 13.5 volts. I am sorry it was my mistake that I said it is 5 volts. Would 13.5 volts input voltage can cause issues ? 

However, I am confused how the input power (voltage) can cause memory issue in chip. Here is my output on terminal while programming Due board. 

Set binary mode
readWord(addr=0)=0x20001000
readWord(addr=0xe000ed00)=0x412fc230
readWord(addr=0x400e0740)=0
readWord(addr=0x400e0940)=0x285e0a60
version()=v1.1 Dec 15 2010 19:25:04
chipId=0x285e0a60
Connected at 921600 baud
readWord(addr=0)=0x20001000
readWord(addr=0xe000ed00)=0x412fc230
readWord(addr=0x400e0740)=0
readWord(addr=0x400e0940)=0x285e0a60
Atmel SMART device 0x285e0a60 found
write(addr=0x20001000,size=0x34)
writeWord(addr=0x20001030,value=0x40)
writeWord(addr=0x20001020,value=0x20010000)
writeWord(addr=0x400e0a00,value=0x600)
writeWord(addr=0x400e0c00,value=0x600)
Device       : ATSAM3X8
readWord(addr=0)=0x20001000
readWord(addr=0xe000ed00)=0x412fc230
readWord(addr=0x400e0740)=0
readWord(addr=0x400e0940)=0x285e0a60
Chip ID      : 285e0a60
version()=v1.1 Dec 15 2010 19:25:04
Version      : v1.1 Dec 15 2010 19:25:04
Address      : 524288
Pages        : 2048
Page Size    : 256 bytes
Total Size   : 512KB
Planes       : 2
Lock Regions : 32
Locked       : readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x400e0a04,value=0x5a00000a)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
readWord(addr=0x400e0a0c)=0
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x400e0a04,value=0x5a00000a)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
readWord(addr=0x400e0a0c)=0
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x400e0a04,value=0x5a00000a)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
readWord(addr=0x400e0a0c)=0
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x400e0a04,value=0x5a00000a)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
readWord(addr=0x400e0a0c)=0
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x400e0a04,value=0x5a00000a)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
readWord(addr=0x400e0a0c)=0
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
[... 179 lines cut ...]
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x400e0a04,value=0x5a000005)
readWord(addr=0x400e0a08)=0
readWord(addr=0x400e0c08)=0x1
readWord(addr=0x400e0a08)=0
readWord(addr=0x400e0c08)=0x1
readWord(addr=0x400e0a08)=0
readWord(addr=0x400e0c08)=0x1
readWord(addr=0x400e0a08)=0
readWord(addr=0x400e0c08)=0x1
readWord(addr=0x400e0a08)=0
readWord(addr=0x400e0c08)=0x1
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x1
writeWord(addr=0x400e0c04,value=0x5a000005)
done in 0.013 seconds

Write 24392 bytes to flash (96 pages)

[                              ] 0% (0/96 pages)write(addr=0x20002000,size=0x100)
writeWord(addr=0x20001028,value=0x80000)
writeWord(addr=0x2000102c,value=0x20002000)
readWord(addr=0x400e0a08)=0x1
readWord(addr=0x400e0c08)=0x5

pdborhad, 2017-11-10:
Hi,

Have anybody come across this issue before? 

Regards,
P