arduino/ArduinoCore-sam #50: wiring_constants.h define of true/false causing trouble

thinkyhead, 2018-05-12:
The `wiring_constants.h` file defines `true` as `0x1` and `false` as `0x0` and this conflicts with the definition in `stdbool.h` included by `Arduino.h` (and others). The compiler bails when it hits this issue, causing builds to fail. For example…

- https://travis-ci.org/MarlinFirmware/Marlin/builds/378058672#L1537

How can this possibly be avoided or fixed?

thinkyhead, 2018-05-12:
Seems to be something else. These aren't the primary cause.

closed 2018-05-12 by thinkyhead

bblanchon, 2025-06-19:
Replacing `true` and `false` with `1` and `0` changes their type from `bool` to `int`, which breaks the overload resolution.
See https://github.com/bblanchon/ArduinoJson/issues/2181#issuecomment-2987343833 for an example.

Please reopen this issue.