arduino/ArduinoCore-sam #62: PluggableUSB.h location inconsistency

gtrainavicius, 2018-11-06:
AVR arduino has PluggableUSB.h in 'arduino' subfolder, while the ARM based cores (SAMD including) have it in 'USB' subfolder, making it difficult to have a library which supports both cores.

You may fix this inconsistency by placing a 'dummy' PluggableUSB.h in the ARM cores in arduino/ subfolder which simply includes the USB/PluggableUSB.h, or do the reverse of this in the AVR core.

gtrainavicius, 2018-11-06:
Actually upon closer inspection, the API itself is defined in an incompatible way, however, they are still very similar (some constants and types renamed, different storage type for endpoint type (uint32_t vs uint8_t)) and it would have made a lot of sense to have the same USB API for boards that have USB peripheral support, available across Arduino cores...

Any recommendations what's the best way forward to maintain a USB library that supports PluggableUSB on AVR and ARM architectures? Specialized implementations for each core?

sylvainpelissier, 2018-11-06:
For the MIDIUSB project they used `#if defined(ARCH)` macro: https://github.com/arduino-libraries/MIDIUSB/blob/master/src/MIDIUSB.h
That's what I did for my projects but it is difficult to maintain.

Maybe including the defintions in a 'dummy' PluggableUSB.h can be simpler for future projects.

facchinm, 2018-11-08:
Hi @gtrainavicius ,
the effort of https://github.com/arduino/ArduinoCore-API is to provide a standardized way to handle this; once adopted, every supported platform will need to include `api/PluggableUSB.h` (for example usages, see that project's README).
In the meantime the only supported way is changing the include path via ifdefs, sorry.

gtrainavicius, 2020-01-14:
Hey, has there been any new developments on adopting a consistent API for PluggableUSB.h?