arduino/ArduinoCore-sam #69: error: this 'else' clause does not guard... [-Werror=misleading-indentation]

Lauszus, 2019-08-08:
Due to the mix of spaces and tabs used for indentation the following lines: 
https://github.com/arduino/ArduinoCore-sam/blob/0a4c3b196a02e48e31b752a05d8c8064007874dc/cores/arduino/USB/USBCore.cpp#L688-L690 are causing the following warning:
```
/home/travis/.platformio/packages/framework-arduinosam/cores/sam/USB/USBCore.cpp: In function 'void USB_ISR()':
/home/travis/.platformio/packages/framework-arduinosam/cores/sam/USB/USBCore.cpp:688:21: error: this 'else' clause does not guard... [-Werror=misleading-indentation]
                     else
                     ^~~~
/home/travis/.platformio/packages/framework-arduinosam/cores/sam/USB/USBCore.cpp:690:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'else'
         UDD_Send8(EP0, 0);
         ^~~~~~~~~
```

See: https://travis-ci.org/felis/USB_Host_Shield_2.0/jobs/569237654

A fix would simply be to replace all tabs with spaces or vice versa.

arduinotech, 2020-02-04:
Fix please

Suraj-singh999898, 2020-03-03:
Did it resolve?

Lauszus, 2020-03-04:
@Suraj-singh999898 no: https://github.com/arduino/ArduinoCore-sam/blob/master/cores/arduino/USB/USBCore.cpp#L689-L690

Lauszus, 2020-11-15:
Seems like this was fixed in: https://github.com/arduino/ArduinoCore-sam/commit/790ff2c852bf159787a9966bddee4d9f55352d15, so closing this issue.

closed 2020-11-15 by Lauszus