arduino/ArduinoCore-sam #83: USB DUEr3 Host powering cutoff with ext pwr >7Vs

neilh10, 2019-09-16:
If the USB port is  to be used as HOST  and is to supply power, then there needs to be an external power supply capable of delivering the 0.5A on the USB Host +5V.
The LM2734 is capable of doing that from 7-12V
However if the input voltage is >7V it switches off the USB Host voltage supply.
I think it should support the full range of input voltage.

Analysis: It seems to me the IC1B/LMV3518 +/- inputs are switched around (easy to do - I've got +/- wrong way round before)
I verified it working with the DUE hardware for Vin < 7V. 
Workaround: There is a software fix for disabling the UOTGVBOF hw processing so that a Vin>7V can be used, but its a kluge and OTG isn't then supported. 
Other messages that have discussed this
http://arduino.cc/forum/index.php/topic,140839.0.html
http://arduino.cc/forum/index.php/topic,135399.0.html
http://arduino.cc/forum/index.php/topic,146574.msg1173826.html#msg1173826
It seems to me the "Taijiuino_Pro_ArduinoDue_V1.1" has an approach that works (but I haven't tested the hw), however it doesn't detect in hw if the input drops below 7V

sandeepmistry, 2019-09-16:
Hi @neilh10, have you had a chance to try @cmaglie's patch above?

neilh10, 2019-09-16:
long time ago... I stopped using the DUE. so I didn't check it

closed 2019-09-16 by neilh10