arduino/ArduinoCore-sam #97: Arduino Due USB Implementation Error

RaceCore24, 2019-12-16:
Arduino=SAM v1.6.12
The problem occurres when you have a Windows 10 application retrieving custom HID reports in high tempo from the Due. After approx 02:05 minutes every hid report returnes correct reportId but only zero values in the other report data fields. 

I have already ruled out the fw from my perspective. The windows application continues to run but with these reports zeroed out. In the windows application everything seems fine. The windows application runs fine on other boards. The USBD_xxx methods seems unfinished.

Just to add. In windows the hid device continues working as expected for some time after this. But then finally stops working as well. If I do not run the windows application polling hid reports, then the Due never fails.

Any tips or ideas as to what might be the problem would be greatly received.

RaceCore24, 2019-12-19:
Finallly got to debug the Due. The problem is that this line:
UDD_Recv(EP0, (uint8_t*)&setup, 8); 
in
USBCore.cpp
in
USB_ISR

Receives all 0's in setup from the UDD_Recv.

Tomorrow I will look more into it. But there is something wrong with the USB stack for SAM.

billydragon, 2020-12-24:
I think i have the same problem when Due running force feedback USB HID.  My code is run fine on Leonardo. The code is same compiled on Due with no errors and device run normal except wrong USBD_recv data, just 2 bytes insteed of 64. Are you have solution to fix?

sith-ikjetil, 2021-01-23:
> I think i have the same problem when Due running force feedback USB HID. My code is run fine on Leonardo. The code is same compiled on Due with no errors and device run normal except wrong USBD_recv data, just 2 bytes insteed of 64. Are you have solution to fix?

No I have not solved the problem. I have been busy with other things. But this is an annoying problem. I would love for the due to work properly. Im gonna look again at the problem and see if I can come up with some fix.

I also have code on other microcontrollers running fine, but on due fails. It might have something to do with USB 2.0 High Speed contra the other microcontrollers full speed.

If you (billydragon) have solved the problem please share solution.

billydragon, 2021-01-23:
im not sure, but after fix this, my FW for FFB joystick work with received data.
uint32_t USBD_Recv(uint32_t ep, void* d, uint32_t len)
{
    if (!_usbConfiguration)
        return -1;

    LockEP lock(ep);
    //uint32_t n = UDD_FifoByteCount(ep & 0xF);
    //len = min(n,len);
    //n = len;
    //uint8_t* dst = (uint8_t*)d;
    //while (n--)
        //*dst++ = UDD_Recv8(ep & 0xF);
	len = min(UDD_FifoByteCount(ep & 0xF),len);
        uint8_t* dst = (uint8_t*)d;
        UDD_Recv(ep & 0xF, dst, len);
    if (len && !UDD_FifoByteCount(ep & 0xF)) // release empty buffer
        UDD_ReleaseRX(ep & 0xF);

    return len;
}

sith-ikjetil, 2021-01-23:
>im not sure, but after fix this, my FW for FFB joystick work with received data.

Unfortunately it did not work for me. I believe I have the opposite problem. Sending seems to be the problem for me. I will continue to look at it.