arduino/ArduinoCore-sam #98: SPI pins

iddq, 2020-01-18:
SPI0_NPCS0 is PA28 and that is PIN 111 on the 144 pins package

Why is it 77 here?

https://github.com/arduino/ArduinoCore-sam/blob/eed66e7977b4037f49bbcd832437d4bcf72c0f3f/variants/arduino_due_x/variant.h#L108

per1234, 2020-01-18:
77 is the Arduino pin number. This is just an arbitrary number used to refer to the pins in Arduino IO functions (e.g., `pinMode`, `digitalRead`, `digitalWrite`). Arduino pin numbers don't necessarily have any relation to the physical pin number.

iddq, 2020-01-18:
Thanks. Can I find a schematic that uses this numbering? Can you please tell me which pin is used for SPI CS (Chip Select) by default if I'm not providing any pin number at SPI.begin() ?

per1234, 2020-01-18:
The Due's [schematic ](https://content.arduino.cc/assets/DUE_V02g_sch.pdf)and [pin mapping table](https://www.arduino.cc/en/Hacking/PinMappingSAM3X) are both available on the "Documentation" tab of the Due's product page:
https://store.arduino.cc/arduino-due

But this isn't the appropriate place for this sort of discussion. This issue tracker is meant to be used only for reporting bugs and submitting feature requests. For general assistance, the appropriate place is the [Arduino Forum](https://forum.arduino.cc/). We're happy to provide all the assistance you need over there.

closed 2020-01-18 by iddq