arduino/ArduinoCore-sam #109: using Serial.printf to print uint64_t

trlafleur, 2020-08-05:
If I try to use Serial.printf to print two uint64_t variables, the 2nd one prints as a zero...

Arduino 1.8.12
Feather M0
Adafruit SAMD21 Boards 1.6.0

~~~
void setup() {
     Serial.begin (115200);              // Initialize USB/serial connection
     delay(2000);

     Serial.printf("hello...\n");

uint64_t A = 12345678l;
uint64_t B = 87654321l;

Serial.printf ("A: %lu, B; %lu\n", A, B);

Serial.printf ("A: %lu\n", A);
Serial.printf ("B; %lu\n", B);

}

void loop() {
  // put your main code here, to run repeatedly:

}
~~~

matthijskooijman, 2020-08-05:
I don't think this core offers `Serial.printf`, I suspect that's an Adafruit addition? In any case, since you're using an adafruit board and (presumably) also their core, this report should probably be left at https://github.com/adafruit/ArduinoCore-samd

I'm going to go ahead and close this, feel free to leave more comments if needed, we can always reopen.

closed 2020-08-05 by matthijskooijman

trlafleur, 2020-08-05:
I moved it to ADAfruits site, but printf support in arduino IDE has been requested many-many times going back 5 or 6 years... don't you think its time to support this very common C and C++ standard feature!!!!! especially with 32bit CPU's like M0 and M4!!

matthijskooijman, 2020-08-05:
That seems to be a different discussion, that is not quite the subject of this issue, so I'll get into that discussion here (or rather not at all, really).