arduino/ArduinoCore-sam #110: adc 12 bit

iddq, 2020-09-11:
Why is it converted to 10 bit if the next line assumes it is 12 bit wide.

https://github.com/arduino/ArduinoCore-sam/blob/eed66e7977b4037f49bbcd832437d4bcf72c0f3f/cores/arduino/wiring_analog.c#L115

matthijskooijman, 2020-09-11:
At first glance, seems like a bug to me. I thought that maybe `adc12b_get_latest_value` would return a zero padded value, but it seems to just return the `LCDR` register as-is:

https://github.com/arduino/ArduinoCore-sam/blob/c893c62ec9953ed36a256b5243272fda2e473c14/system/libsam/source/adc12_sam3u.c#L243-L246

And that register contains the 12-bit ADC value in the lower 12 bits:

[image]

(Note that the upper 4 bits *can* contain the channel number, but only if enabled, so I suspect this is not the case here and these are always zero and can be ignored).

Do you happen to have a Due you could test if this is indeed broken? Should be easy, if all `analogRead` values for these pins are off by a factor four, then it is really a bug.

iddq, 2020-09-11:
I can't test it. I don't have DUE with SAM3U4E