arduino/ArduinoCore-sam #112: Upgrade gcc toolchain version

matthijskooijman, 2020-09-29:
This core still uses the rather old gcc 4.8 toolchain, which lacks a lot of useful (C++) functions.

I just tried replacing it with SAMD's gcc 7 version, which seems to work just fine with a few test sketches I tried. Maybe this would be an easy improvement: Just replace the gcc version with the one from SAMD.

ikcalB, 2025-08-07:
**TL;DR:** _please_ update toolchain. Some fixes and lots of features missing!

Even microchip (ATMEL) themselves have upgraded to 6.3.1.

More than 11 years later, it should not be too much to ask, to update the toolchain...

chepo92, 2025-08-07:
What is the first rule of engineering? If it works just leave it as is, haha, just kidding.
Agree, but if the toolchain is updated there may be a lot of things that may break (and have been working all this time), also needs to be tested in many devices, maybe someone can make a dev/test branch that some people can test